BrainFPV/betaflight #1: Board Doesn't Save Port/Receiver Settings

swissfreek, 2017-01-07:
Flashed RC4 and my board doesn't seem to retain port or receiver settings on reboot. It always resets the ports page to defaults (basically everything off), and reset the receiver type to PPM RX.

Other settings do stick, such as OSD and other items on Configuration page (features, gyro update rate, craft name, etc.).

The board is uninstalled and just connected to the computer, not connected to any other components. It was flashed to RC4 using the dRonin GCS.

mluessi, 2017-01-07:
@swissfreek I just tested it; works as expected for me. Please double check. If it doesn't work, what version of the configurator are you using and what port settings are you trying to set?

swissfreek, 2017-01-07:
OK, I will try again when I get back from this business trip next weekend. I'll be building the actual copter at the same time so maybe things will change when I have all the peripherals connected. Totally willing to believe it's me.

The specific settings that wouldn't take for me is I was trying to set SerialRX on UART6, set UART1 to SmartAudio, and then on the Configuration page setting receiver type to Serial and CRSF as the receiver protocol. All of these settings revert to default on board reboot. Every other setting I've changed sticks, though.

RoadyFPV, 2017-01-08:
Hi i have the same issue with the BrainRE1. I must try it 3 or 4 times press "save n reboot".
But only on pages with the button "save n reboot".

The log in configurator say this when i press save n reboot:
2017-01-08 @ 14:52:04 -- Unrecoverable failure of serial connection, disconnecting...
2017-01-08 @ 14:52:04 -- Serial port successfully closed

Sorry, my english is not so good. I'm from germany.

mluessi, 2017-01-08:
@RoadyFPV on the ports page, make sure that for UART1 MSP is disabled when selecting SmartAudio. If it is enabled, it will be set back to defaults. There was a user on the facebook group who had this problem.

Athir1, 2017-01-12:
hi there 
is this can be fix ? https://www.youtube.com/watch?v=YmtAmckCxNA

closed 2017-02-19 by mluessi