BrainFPV/betaflight #2: cant connect my fc

Athir1, 2017-01-13:
hi there
is this can be fix ? https://www.youtube.com/watch?v=YmtAmckCxNA

RoadyFPV, 2017-01-15:
Hi 

maybe try this from BrainFPV Blog.

Important: Never select "Full chip erase" when flashing Betaflight. Doing so will erase the dRonin bootloader. If you accidentally did this, download dRonin and use the above steps to flash the file "ef_brainre1.hex" (located under "firmware") onto RE1. After this, do the steps again to flash Betaflight. If you can't connect to Betaflight and want to erase the settings, use the "Safe Boot" method described below.
2: Flashing using the dRonin GCS

This method use the dRonin GCS and the dRonin bootloader to flash Betaflight on RE1. It is best used if you have already installed your RE1 on your multirotor and can't access the BOOT pads on the bottom. The dRonin GCS can also be used to load Betaflight in "Safe Boot" mode, which means all settings are set back to defaults. To flash your RE1 with Betaflight do the following:

    Download the latest "betaflight_*_BRAINRE1.tlfw" file from the BrainFPV Betaflight GitHub.
    Download dRonin and start the GCS.
    Disconnect RE1 from USB and power.
    In the dRonin GCS, go to the "Firmware" tab and click on "Rescue".
    Connect RE1 to USB. The dRonin GCS should connect to the bootloader and the green LED on RE1 will be pulsating slowly.
    Click on "Open File.." and select the "betaflight_*_BRAINRE1.tlfw" file. Note: you will need to adjust the file type filter to show all.
    Click on "Flash"
    Once the flashing is finished, you can click on "Boot" to start Betaflight.

Using Safe Boot: To reset all Betaflight settings to default, click Rescue (step 4 above), connect RE1 to USB and then click on "Safe Boot". This can be useful if the Betaflight settings become corrupted for some reason.

Greetings

Athir1, 2017-01-15:
Thanks for replying I just fixed last night
Many thanks

---- RoadyFPV wrote ----

Hi

maybe try this from BrainFPV Blog.

Important: Never select "Full chip erase" when flashing Betaflight. Doing so will erase the dRonin bootloader. If you accidentally did this, download dRonin and use the above steps to flash the file "ef_brainre1.hex" (located under "firmware") onto RE1. After this, do the steps again to flash Betaflight. If you can't connect to Betaflight and want to erase the settings, use the "Safe Boot" method described below.
2: Flashing using the dRonin GCS

This method use the dRonin GCS and the dRonin bootloader to flash Betaflight on RE1. It is best used if you have already installed your RE1 on your multirotor and can't access the BOOT pads on the bottom. The dRonin GCS can also be used to load Betaflight in "Safe Boot" mode, which means all settings are set back to defaults. To flash your RE1 with Betaflight do the following:

Download the latest "betaflight_*_BRAINRE1.tlfw" file from the BrainFPV Betaflight GitHub.
Download dRonin and start the GCS.
Disconnect RE1 from USB and power.
In the dRonin GCS, go to the "Firmware" tab and click on "Rescue".
Connect RE1 to USB. The dRonin GCS should connect to the bootloader and the green LED on RE1 will be pulsating slowly.
Click on "Open File.." and select the "betaflight_*_BRAINRE1.tlfw" file. Note: you will need to adjust the file type filter to show all.
Click on "Flash"
Once the flashing is finished, you can click on "Boot" to start Betaflight.

Using Safe Boot: To reset all Betaflight settings to default, click Rescue (step 4 above), connect RE1 to USB and then click on "Safe Boot". This can be useful if the Betaflight settings become corrupted for some reason.

Greetings

closed 2017-02-19 by mluessi