BrainFPV/betaflight #4: Erasing dataflash just hangs?

joeynuggetz, 2017-01-22:
When I try to erase the blackbox dataflash via BF configurator, it completes but the progress indicator just sits there forever. Once I disconnect and reconnect then dataflash shows empty so it seems to work but there seems to be a GUI problem.

Windows 10
Running BTFL 3.1.0 on configurator 1.9.0

RoadyFPV, 2017-02-19:
I have the same issue. 
Win10; BF 3.1.5; 1.9.1 Configurator

Greetings

mluessi, 2017-02-19:
Yes, this is a known issue. I will will look into fixing it.

cbehrends, 2017-05-12:
I upgraded to 3.1.7 a couple of weeks ago and this issue has been resolved.  It takes about 15-20 seconds to erase the data flash on my RE1.

mluessi, 2017-05-14:
Ah yes, this was fixed in 3.1.7 but I forgot to post it here. Thanks for reporting.

closed 2017-05-14 by mluessi