BrainFPV/betaflight #5: Feature Request:  Better BetaFlight OSD support

cbehrends, 2017-02-03:
As a OSD user, it would be cool if I had full Betaflight control over the OSD layouts. 
While the current feature set is nice for sure, some options such as adding/removing the various horizon bars simply do not function in Betaflight.  It would be nice to have just the Crosshairs and the single artificial horizon without the graduated incline bars.

My FatShark Teleporters get hard to view the OSD with too many things happening on the screen.

Great product, I love where this is going.  (a Piko version would be cool for my 90mm build ;)

cbehrends, 2017-02-05:
@mluessi I'm reading up on the OSD code and may take a shot at extending it / enhancing it.  If I have any success I'll let you know.

mluessi, 2017-02-05:
@cbehrends the OSD is fully configurable using the configurator. The AHI just doesn't have the bars because it's not needed IMO. You can enable the center mark; it's called crosshairs.

In general, I don't think we should add a lot of OSD features that are not present in regular betaflight. I would make the fork harder to maintain and we would need a separate version of the configurator.

cbehrends, 2017-02-05:
@mluessi It may be that I'm simply mistaken, I'll take another look.  But to me it appeared that the behavior I'm describing is a feature of regular Betaflight that the RE1 doesn't fully support.  I agree that there should not be any non standard Betaflight elements in the OSD.  At the same time, it would be nice if the RE1's behavior matched that of the Betaflight configurator.

Thanks for responding!  Reading the code has actually been pretty easy...it's well written.

cbehrends, 2017-02-05:
A little more research and I think I see where my confusion lies.  The RE1 is not using a 'standard' Betaflight OSD, but a vector graphics one (which is bad ass by the way).  The OSD I'm used to in Betaflight does not quite look the same as the Taulabs/Dronin one that this implementation uses.  That being said, all of the elements are there and working fine.  It appears it would be a matter of matching the Betaflight look/feel and separating the toggling of the 'Horiz' and 'Sidebar' elements.  Just for nerdiness, I'm going to take a stab at it.  My OCD side screams at me when my layouts don't match :D 

I respect your opinion that you don't feel the sidebar feature is needed and will not submit any PR's.

Thanks again for a great product!

closed 2017-04-25 by mluessi