BrainFPV/betaflight #6: Passive buzzer support?

kh4, 2017-02-19:
I just reflashed from dRonin and sorted out motor ordering et all, Everything is working but it seems BF version is lacking support for driving a passive buzzer?? or I am simply blind.

mluessi, 2017-02-19:
Do you mean a buzzer that needs a 4kHz AC drive? That's correct, very few people use these buzzers so I didn't add it initially, but I can add it.

kh4, 2017-02-19:
Yeah, exactly. I happen to have some buzzers salvaged from cheap Lipo voltage alarms (like the HK 2-8S one) which need a 4kHz (or so) signal to drive them.

It would be nice to have direct support for them.

kh4, 2017-02-19:
Ah I can see it is really easy to add for RE1 since it is just one bit in FPGA config reg, coolness ;)

kh4, 2017-02-20:
Hacked it up and compiled a firmware for AC style buzzer.

https://github.com/kh4/betaflight/commit/184fa9343f8db57b9516906b2680d263c6cc207d

Firmware binaries if someone wants to use this too

http://kari.hautio.net/brain/betaflight_3.1.5_BRAINRE1-ACBZ.hex
http://kari.hautio.net/brain/betaflight_3.1.5_BRAINRE1-ACBZ.tlfw

kh4, 2017-02-20:
I guess this is about the proper way to do it, it would be only in CLI though. I have not tested this though as I only have one brainRe1 board.

kh4, 2017-02-20:
Yes, it makes sense to have it in RE1 config. I was also looking for a way to avoid bumping EEPROM.

If using RE1 config it makes sense to set it in FPGA init which avoids changing any beeper code.

There is no hurry as I am happily flying with the modified firmware mentioned in earlier commit.

mluessi, 2017-02-20:
Sounds good. It should be fine without bumping the EEPROM version, as it uses a CRC and adding a new field to the config pretty much guarantees that the CRC check will fail and the config will be initialized with defaults.

closed 2022-01-16 by github-actions[bot]