BrainFPV/betaflight #8: BF 3.1.6 JETIEXBUS

RoadyFPV, 2017-03-02:
How to conect JETIEXBUS to UART3. And how can i disable Bidirectional mode?
I do not want to break anything.

thx RoadyFPV

mluessi, 2017-03-02:
It seems like the JETIEXBUS opens the UART in RXTX mode, which isn't supported on UART3 on RE1. Did you have it working with earlier versions? I added code to prevent this type of configuration on UART3 as it isn't supported, but it seems like the JETIEXBUS hangs as it doesn't check for errors.

RoadyFPV, 2017-03-02:
I fly BF 3.1.5 with JETIEXBUS, this works fine. But when i select JetiEXBus in 3.1.6  than i cant connect to re1.

mluessi, 2017-03-02:
OK, I realized that the check I added isn't actually needed. Please try this version: https://github.com/BrainFPV/betaflight/releases/tag/v3.1.6-BRAIN-2

RoadyFPV, 2017-03-02:
Thx for ur fast help. I will try it tomorrow and give a feedback.

RoadyFPV, 2017-03-03:
Hi mluessi,

i try the new version. But when i select SerialRX and JetiexBus i cant connect to re1. Plug in USB and no com port detect.

RoadyFPV, 2017-03-03:
I see when i load the new firmware in dronin it say 3.1.6Brain1 but i select your new file Brain2

RoadyFPV, 2017-03-05:
@mluessi 
have u n idea why i cant connect to re1 after i select jetiexbus? i flashd the brain2 again but with the same result. wen i selct sbus i can connect. but is the wrong protocol

mluessi, 2017-03-08:
Can you try 3.1.6-BRAIN-2 again? It's possible that the .tlfw file wasn't rebuilt properly and I just uploaded it again.

RoadyFPV, 2017-03-08:
I loaded the .tlfw file again and flashd with the same result. when i select jetiexbus i cant connect to re1.
I plugd in usb windows detect usb with sound but the second detect sound will not come. after this no com port detect and windows will bring a massage that the usb device is not workd correctly.

RoadyFPV, 2017-03-12:
Hi 
i try to flash the betaflight hex file on my re1 because the tlfw file wont work. but without success. is it even possible to flash hex file on re1?

RoadyFPV, 2017-03-15:
@mluessi 
can u pls upload the new 3.1.6 tlfw file for me? i  would like to use 3.1.6 but i cant.

mluessi, 2017-03-15:
@RoadyFPV I'm looking into it; there were changes introduced in Betaflight 3.1.6 that are causing problems with RE1. Namely, JETIEXBUS configures the serial port to also send data to the receiver, which isn't supported UART3 on RE1. For now, use either 3.1.5 or connect the receiver to UART1.

I will let you know once I have a fix that you can test. Later this week, hopefully.

RoadyFPV, 2017-03-15:
ok
thank you very much. but i use tbs smartaudio on uart1. so i will wait and use 3.1.5. enter the osd menu from stats screen is very nice in 3.1.6.

mluessi, 2017-03-16:
@RoadyFPV please try this version: https://drive.google.com/open?id=0ByA776HMnh4gRERUT1ZDb3hrSUk

RoadyFPV, 2017-03-16:
@mluessi
this works, i can use jetiexbus. testflight come later, configuration is fine. thank you.

RoadyFPV, 2017-03-17:
@mluessi 
Test flight was successful. Thanks for ur great work i'am so happy.
Greetings

closed 2017-04-25 by mluessi