BrainFPV/betaflight #9: Firmware

dane2992, 2017-03-26:
I don't know why I would be having this issue but any help is appreciated. So using dRonin to flash the betaflight firmware to the re1. when I select open file to select the firmware, the firmware isn't showing. ive changed setting while saving the file and also changed he locations. I'm at a lose, ive followed the directions on the website and also a video on youtube.

ragd011, 2017-03-26:
Are you using the tlfw file?  That one is for dronin gcs based installs.

dane2992, 2017-03-26:
Yes I am. I've saved it in different locations as well as saved it as all file types too.

ragd011, 2017-03-27:
I may be confusing my cc3d.  I'll check.  I may have just loaded the dfu drivers and from cli entered dfu to load the hex.

dane2992, 2017-03-27:
I'm trying to get help from all over the web, even on rcgroups thread specifically on the re1

mluessi, 2017-03-27:
It think you figured it out (via Facebook), correct?

dane2992, 2017-03-27:
Yes sir that was me. Making sure your on the correct acct on the computer helps (even tho it's my personal, was on administrator acct)

closed 2017-03-27 by dane2992