BrainFPV/betaflight #15: Can't flash FW via BF Configurator

RoadyFPV, 2017-08-18:
Hi,

i try to flash BF3.2 to my RE1 but cant flash. Failed to open Serial Port.
I install Zadig now the bootloader shows COM8 in BF Configurator but when i hit Flash comes Failed to open Serial Port. What am I doing wrong.
Thanks for help

closed 2017-08-18 by RoadyFPV

mluessi, 2017-08-18:
When flashing 3.2 the first time and BF 3.1.7 (or older) you need to click "Flash" twice. The first time, it will reboot RE1 and a second or two later a "DFU" device should show up. When it does, click "Flash" again and it will flash it. This is due to the presence of the dRonin bootloader, which introduces some delay in the reboot process. After flashing 3.2, it won't have the delay and should flash like any other BF flight controller.

Is this what fixed your problem?

RoadyFPV, 2017-08-18:
The problem is fixed.
I try all treiber with Zadig. One of it shows DFU mode then i coud flash bf 3.2
Thank you