BrainFPV/betaflight #18: mAh Draw on Taranis only shows Capacity i've entered into the Power and Battery Tab

idiotsniff, 2017-11-14:
Hi there, 

I'm having issues with betaflight 3.2.1 on a Brainfpv RE1 board not showing the mAh used on my Taranis X9D plus, it will only populate whatever battery capacity in the Power & Battery Tab in the BF configurator. if I set it for 1200mAh capacity it shows 1200 on my taranis under the "capa" abbreviation on my crossfire setup. have never had luck making this work and have been having issues lately with erratic voltages all over the place and mAh and Current readings going up into the thousands. I have Voltage and Current scales as they shuold be the 66 voltage and 250 for current. don't know how to make the radio show whats being used rather then just telling me what the battery capacity I've typed in.

hopefully its just something simple?

thanks for your time,

idiotsniff

# If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

## Please consider using other user support options such as asking the manufacturer of the hardware you are using, [RCGroups](https://rcgroups.com/forums/showthread.php?t=2464844), [Slack](https://slack.betaflight.tech) or other user support forums & groups (e.g. facebook).

## If you believe there is an issue with the firmware itself please follow these steps:
1.) Describe your problem;
2.) Include ways to reproduce the problem;
3.) Provide information about your flightcontroller and other components including how they are connected/wired;
4.) Add the used configuration and firmware version;
5.) Create a `diff` and post it here in a code block. Put \`\`\` (three backticks) at the start and end of the `diff` block; and
6.) Remove this Text :).

mluessi, 2017-11-14:
If the voltage / current readings are wrong in the OSD, it's most likely a hardware problem. Please contact support@brainfpv.com to open a support case.

closed 2017-11-14 by mluessi

idiotsniff, 2017-11-14:
hi there, voltage is correct, current draw is fine like the live amperage
usage, but the mAh draw just populates whatever I type into the battery
size warnings box in betaflight. so if I put 0, it reads 0, if I put 1000,
then it reads 1000 in the Taranis telemetry info.

is there anything else that needs to be done to make it show the correct
reading? do I have to put in a calculation or any other setting under the
capa telemetry sensor settings in the Taranis?

thanks for your help,

Joel Brown