BrainFPV/betaflight #21: Start Runcam Split 2 record restart Board and ESC'S

RoadyFPV, 2018-03-19:
When i start the record on RCS2 via AUX Chanel then the Radix restarts Beeper Beeps and ESC restart.
Start recort via Button on RCS2 Board is fine.

mluessi, 2018-03-22:
@RoadyFPV what version are you using? Also, please paste your config (type "diff defaults" in the CLI)

mluessi, 2018-03-22:
Ah, I see there is an open BF issue https://github.com/betaflight/betaflight/issues/5398

RoadyFPV, 2018-03-22:
Yes is see it too and confirmed it.

RoadyFPV, 2018-04-15:
Hi i recived an new fw for the split 2 from runcam. I contact RC when no WIFI is conect and select WIFI over the MODE Tab the Split show black screen and shut down.
With the new FW only massage WIFI ERROR shows in the screen. But the best is, the ESC dont reset when i start recording via MODE TAB (AUX Chanel).

I cant test more in the moment while my TX is in service.

closed 2022-01-16 by github-actions[bot]