BrainFPV/betaflight #25: High CPU Load (46% Idle) on betaflight_v3.4.0-RC3-BRAIN-1_RADIX

th3whit3f, 2018-06-26:
Hello,

i have update the radix FC from 3.3.3 to 3.4.0 RC2 and RC3 but then the CPU Load is getting high on 3.4.0 firmware about 30% till 50%.
It's keep getting stuck on betaflight configurator, when i save and reboot.
The Radix Fc is right out of the box, my second radix is on my quad and has the same problem.
The radix fc are both one week old.
when downgrade to 3.3.3 there is no issue, with the cpu load and getting stuck on betaflight.

My Dump File of the 3.4.0 RC3
[Radix 3.4.0 RC3.txt](https://github.com/BrainFPV/betaflight/files/2138886/Radix.3.4.0.RC3.txt)

Setup.
Radix FC
Radix PDB
Crossfire Nano 12ch
Tramp HV v2 Int version.
Kiss Esc 24a last firmware telemetry on
T-Motor F40 III 2400KV
Runcam Swift 2

giova-fr, 2018-08-11:
Same problem with RE1 on 3.4.0 stable release.
Accelerometers take around 20%

th3whit3f, 2018-08-13:
yep the same problem on 3.4.0 stable, 
please fix the problem and keep your firmware uptodate,

brainfpv have to keep up with the release firmware form betaflight github

yosemitez, 2019-01-13:
just update to 3.5.4 fresh install no changes
CPU:30%, cycle time: 311, GYRO rate: 3215, RX rate: 33, System rate: 9

closed 2022-01-16 by github-actions[bot]