BrainFPV/betaflight #27: v3.5.4-BRAIN-1 - no osd in Configurator: 10.5.0, works in 10.4, guess we need a fw4?

yosemitez, 2019-01-13:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

Please note that feature requests are not 'fire and forget'. It is a lot more likely that the feature you would like to have will be implemented if you keep watching your feature request, and provide more details to developers looking into implementing your feature, and help them with testing.

mluessi, 2019-01-14:
10.5.0 is not intended to be used with betaflight 3.5. Please use 10.4.0 with BF 3.5.4

closed 2019-01-14 by mluessi

yosemitez, 2019-01-14:
copy that