BrainFPV/betaflight #28: Radix and Configurator Freeze when activating Telemetry

RoadyFPV, 2019-03-30:
By flashing the 4.0 to the Radix (clean flash) connect to the configurator, activate Telemetry hit save and reboot, now the Radix and Configurator freeze. When i unplug the Radix from USB the Configurator works again.

only way to fix, is short the boot pads and flash 4.0 again

FPVRaven, 2019-03-30:
I have same issue, as soon as i enable telemetry and hit save everything freezes.....After that even driver fixer freeze as well, plug in lipo = freeze as well. The led next to the usb become static red and the one after it does blink.
BTW i have a R9 and R9 mini, both flashed with the latest flex firmware (eulbt firmware),  i hope this helps :)

RoadyFPV, 2019-03-31:
I tried it with Crossfire and Jeti ExBus, both with the same result.

mluessi, 2019-03-31:
Thanks for the report. I'm looking into it

zakre256, 2019-04-01:
I had the same issue and I had to jumper the boot pad and reflash the firmware. Tried numerous different things to prevent it but activating telemetry in config, freezes/fake bricks the board. So I left it off and no problems so far

mluessi, 2019-04-03:
This has been fixed in 4.0.0-RC5-BRAIN-1. Please try the latest release

RoadyFPV, 2019-04-04:
Ok the freeze is gone! Thx
But i have an other question. Why outputs the buzzer a permanent tone when u arm and have motor stop enabled?
The old version of the buzz was not so annoying ;)

closed 2022-01-16 by github-actions[bot]