BrainFPV/betaflight #29: Betaflight 4.0 RC6 CPU overlaod

FPVRaven, 2019-04-12:
Hey there, RC6 does not work with RPM Filter, my quad freaked out while try to hover! 
Seems like something overloading the CPU: 

# tasks
Task list             rate/hz  max/us  avg/us maxload avgload     total/ms
00 - (         SYSTEM)     10       2       1    0.5%    0.5%         0
01 - (         SYSTEM)    801    2292       2  184.0%    0.6%        36
02 - (       GYRO/PID)   3206      99      85   32.2%   27.7%      4476
05 - (             RX)     32      76      48    0.7%    0.6%        24
06 - (         SERIAL)     98     118       1    1.6%    0.5%        31
07 - (       DISPATCH)    801       4       0    0.8%    0.0%         7
08 - (BATTERY_VOLTAGE)     49       3       1    0.5%    0.5%         1
09 - (BATTERY_CURRENT)     49      17       1    0.5%    0.5%         0
10 - ( BATTERY_ALERTS)      5       2       1    0.5%    0.5%         0
11 - (         BEEPER)     98      29       1    0.7%    0.5%         1
15 - (      TELEMETRY)    245      96       1    2.8%    0.5%       325
20 - (        VTXCTRL)      5      32      12    0.5%    0.5%         1
21 - (        CAMCTRL)      5       1       1    0.5%    0.5%         0
23 - (    ADCINTERNAL)      2       2       0    0.5%    0.0%         0
24 - (       PINIOBOX)     19       2       0    0.5%    0.0%         0
RX Check Function                   3       2                         1
Total (excluding SERIAL)                       225.2%   32.9%
BrainFPV tasks remaining stacks:
main: 0 Bytes free
idle: 80 Bytes free
Betaflight: 1168 Bytes free
OSD: 712 Bytes free

Running 4K/4K gyro/pid looptimes, also running latest blheli_32 test firmware. Same settings and ESC firmware works fine with 2 other flightcontrollers, got to be some firmware bug. After reflashing the same issue was there, seems like there is no workarround for this one :(

mluessi, 2019-04-16:
I assume it is because of the 4k/4k setting. Please try with having it set to 8k/8k

FPVRaven, 2019-04-16:
Just checked 8k/8k = 
# tasks
Task list             rate/hz  max/us  avg/us maxload avgload     total/ms
00 - (         SYSTEM)     10       2       1    0.5%    0.5%         0
01 - (         SYSTEM)    801    2293       4  184.1%    0.8%        26
02 - (       GYRO/PID)   3206     104      95   33.8%   30.9%      1458
05 - (             RX)     32      90      54    0.7%    0.6%        10
06 - (         SERIAL)     98     117       1    1.6%    0.5%         5
07 - (       DISPATCH)    801       3       1    0.7%    0.5%         2
08 - (BATTERY_VOLTAGE)     49       2       2    0.5%    0.5%         0
09 - (BATTERY_CURRENT)     49      19       2    0.5%    0.5%         0
10 - ( BATTERY_ALERTS)      6       2       1    0.5%    0.5%         0
11 - (         BEEPER)     98      21       3    0.7%    0.5%         0
15 - (      TELEMETRY)    245     112       1    3.2%    0.5%        61
20 - (        VTXCTRL)      6      33       4    0.5%    0.5%         0
21 - (        CAMCTRL)      6       1       0    0.5%    0.0%         0
23 - (    ADCINTERNAL)      4       3       0    0.5%    0.0%         0
24 - (       PINIOBOX)     19       2       0    0.5%    0.0%         0
RX Check Function                   3       2                         0
Total (excluding SERIAL)                       227.2%   36.3%
BrainFPV tasks remaining stacks:
main: 0 Bytes free
idle: 80 Bytes free
Betaflight: 1168 Bytes free
OSD: 728 Bytes free 

Same :/

FPVRaven, 2019-04-17:
I just saw the stable release, however there is nothing fixed, looptimes are wrong, cpu overloaded or the stats in % are bugged if like to log...... there was corrupt log stuff in my older logs as well, could you please start fixing this stuff?! I am allready getting really dissapointed about the firmware for Radix (can`t belive the should be a stable firmware) :( How can it be stable if the tasks are completely wrong? there are older discussions about this as well, these are basic things we need to check for save flying guys, this is really not what i expected........I will try stable release but i have still this = 

# tasks
Task list             rate/hz  max/us  avg/us maxload avgload     total/ms
00 - (         SYSTEM)     10       1       1    0.5%    0.5%         0
01 - (         SYSTEM)    801    2300       1  184.7%    0.5%        17
02 - (       GYRO/PID)   3205      95      81   30.9%   26.4%      1299
05 - (             RX)     32      76      48    0.7%    0.6%         9
06 - (         SERIAL)     98     191       1    2.3%    0.5%         6
07 - (       DISPATCH)    801       6       0    0.9%    0.0%         2
08 - (BATTERY_VOLTAGE)     49       2       1    0.5%    0.5%         0
09 - (BATTERY_CURRENT)     49       2       1    0.5%    0.5%         0
10 - ( BATTERY_ALERTS)      6       2       1    0.5%    0.5%         0
11 - (         BEEPER)     98       2       1    0.5%    0.5%         0
15 - (      TELEMETRY)    245      99       0    2.9%    0.0%        78
20 - (        VTXCTRL)      6      32       4    0.5%    0.5%         0
21 - (        CAMCTRL)      6       1       0    0.5%    0.0%         0
23 - (    ADCINTERNAL)      4       2       0    0.5%    0.0%         0
24 - (       PINIOBOX)     19       1       0    0.5%    0.0%         0
RX Check Function                   2       2                         0
Total (excluding SERIAL)                       224.6%   30.5%
BrainFPV tasks remaining stacks:
main: 0 Bytes free
idle: 80 Bytes free
Betaflight: 1168 Bytes free
OSD: 728 Bytes free

I will try to fly with log, impossible to have that high load, maybe a bug or it will cook my cpu.......

mluessi, 2019-04-17:
It must have something to do with your settings please type "diff defaults" in the CLI in betaflight and post the output here

RoadyFPV, 2019-04-17:
I have the problem too. Clean flash, tasks in cli show every time around 4k. In settings, PID/Gyro 2k/2k...  4k/4k... 8k/8k... 

Tasks shows around 4k on all 3 settings.
I think on 3.5.7 too.

mluessi, 2019-04-17:
@RoadyFPV please post the output of "diff defaults" in the CLI. Without that it's difficult to help.

FPVRaven, 2019-04-17:
Hold on i think i can give the data you need, i know exacly what is causing this and where it is comming from ;) 
I can say that it got nothing to do with the settings, only with logging, but i asume it is a bug since it would not fly at all with 220% cpu load and more, i tested it and it fly, first flip over was caused by changes on the board orientation in config (was able to reproduce this and a friend had it as well).....

mluessi, 2019-04-17:
Please try a different sdcard or re-format your card. I tried your settings with an sdcard inserted and it works fine. I assume it may be trying to read or format the card and keeps failing

FPVRaven, 2019-04-18:
Ok thank you for checking it, allmoast got crazy with it, log was crazy noisy, maybe cpu load was real and pids or filters was not able to work correct......
I got a Kingston 2GB in there, i have only big ones over for testing, i can test samsung evo 32gb, sandisk ultra 32gb and sandisk ultra 64gb, can i find somewhere witch ones are supported for sure? Never had this Problem because i allways had onboard logging before :/

FPVRaven, 2019-04-23:
Finally i found the bug, the shown faliue is wrong, just type in "tasks" for a second time and it will go down to the correct valiue, maybe this is fixabel, if not am at least sure it is just a bug witch shows wrong valiues everytime you type tasks for the first time......

# tasks
Task list             rate/hz  max/us  avg/us maxload avgload     total/ms
00 - (         SYSTEM)     10       2       1    0.5%    0.5%         0
01 - (         SYSTEM)    801    2291       2  184.0%    0.6%       204
02 - (       GYRO/PID)   3206      71      61   23.2%   20.0%      1090
05 - (             RX)     32      68      39    0.7%    0.6%         8
06 - (         SERIAL)     98     107       1    1.5%    0.5%         5
08 - (BATTERY_VOLTAGE)     49       6       2    0.5%    0.5%         0
09 - (BATTERY_CURRENT)     49       1       1    0.5%    0.5%         0
10 - ( BATTERY_ALERTS)      6       6       1    0.5%    0.5%         0
11 - (         BEEPER)     98      14       1    0.6%    0.5%         0
15 - (      TELEMETRY)    245      92       0    2.7%    0.0%        58
20 - (        VTXCTRL)      6      24       3    0.5%    0.5%         0
21 - (        CAMCTRL)      6       1       0    0.5%    0.0%         0
23 - (    ADCINTERNAL)      4       2       0    0.5%    0.0%         0
24 - (       PINIOBOX)     19       5       0    0.5%    0.0%         0
RX Check Function                   2       1                         0
Total (excluding SERIAL)                       215.2%   24.2%
BrainFPV tasks remaining stacks:
main: 0 Bytes free
idle: 80 Bytes free
Betaflight: 1168 Bytes free
OSD: 728 Bytes free

# tasks
Task list             rate/hz  max/us  avg/us maxload avgload     total/ms
00 - (         SYSTEM)     10       1       0    0.5%    0.0%         0
01 - (         SYSTEM)    801       4       2    0.8%    0.6%       211
02 - (       GYRO/PID)   3206      72      60   23.5%   19.7%      1969
05 - (             RX)     32      43      39    0.6%    0.6%        14
06 - (         SERIAL)     98     754       1    7.8%    0.5%         6
08 - (BATTERY_VOLTAGE)     49       6       1    0.5%    0.5%         0
09 - (BATTERY_CURRENT)     49       1       0    0.5%    0.0%         0
10 - ( BATTERY_ALERTS)      5       2       1    0.5%    0.5%         0
11 - (         BEEPER)     98       1       1    0.5%    0.5%         1
15 - (      TELEMETRY)    246       1       0    0.5%    0.0%        59
20 - (        VTXCTRL)      5       1       2    0.5%    0.5%         0
21 - (        CAMCTRL)      5       1       0    0.5%    0.0%         0
23 - (    ADCINTERNAL)      2       2       0    0.5%    0.0%         0
24 - (       PINIOBOX)     19       1       1    0.5%    0.5%         0
RX Check Function                   2       2                         0
Total (excluding SERIAL)                        29.9%   23.4%
BrainFPV tasks remaining stacks:
main: 0 Bytes free
idle: 80 Bytes free
Betaflight: 1168 Bytes free
OSD: 720 Bytes free

My SD card is no issue as well it works, just a confusing bug (was to simple to type in tasks for a second time, so it took me some time to find that out) :D 

Greetings SkyHogRC

closed 2022-01-16 by github-actions[bot]