BrainFPV/betaflight #31: Radix target not appearing in list

evilC, 2019-06-01:
I have a Radix LI and I am attempting to upgrade to BF 4.0.2-BRAIN-1  
When I use the firmware flasher in BF configurator, and select `Load Firmware [Local]`, and point it at `betaflight_v4.0.2-BRAIN-1_RADIX.hex`, I see no Radix target in the list:  
[image]  
Note, for some reason the screen recording puts my cursor in the wrong place, I really am clicking the load local ;)  
Please advise...

OGFPV, 2019-06-01:
It will not show up in the list of "online FW" because it is locally stored on your PC. Just click the "Flash Firmware" button to begin the install.

evilC, 2019-06-01:
Ahh, it looks like it was maybe a glitch in the configurator
Even though I started a fresh configurator, clicked load local, I was still seeing this on that page:  

[image]  
This lead me to believe it was going to flash Omnibus
I can no longer seem to repro this however - now I just see this, which looks OK

closed 2019-06-01 by evilC