BrainFPV/betaflight #32: Core Temp warning after upgrading to BF 4.0.2

evilC, 2019-06-02:
I upgraded my Radix LI BF firmware to 4.0.2 (From 3.5 that it shipped with), and now, shortly after  powering up, I get a core temp warning flashing on the OSD  
Feeling the chip on the FC, it feels hot, so it does not appear to be a false positive.  
I shut it down by the time it hits 80 degrees C (About a minute or two), I dunno how high it will go, I don't really want to find out, not sure how high is safe  
I have disassembled to the point that there is nothing touching the FC any more, just a Spedix IS25 5mm below it  
Running on 3S - it does not appear to happen when only the USB cable is plugged in, can't really tell as I have no video in that instance.  
I dunno if it was running hot previously and whether the warning feature is new or what, but I have been flying it around on 3.5 and it didn't die, so I guess not?

evilC, 2019-06-02:
So I finally plucked up the courage to fly it, and this is a non-issue when flying, it does not exceed 52deg  
However, when on the bench, this is a bit of an issue - I don't really have time to set up OSD, VTX etc before I get the warning.  
Even if I separate out the FC from other components on the bench (ie giving heat maximum chance to dissapate), this is still an issue  
Maybe my FC just has a chip that runs hot - working with PC overclocking, I know not all chips are equal in terms of thermal profile. Do the chips used get graded or tested in this way?

evilC, 2019-07-13:
FYI I got a second Radix LI, and I get exactly the same thing, so it appears that this is an inherent issue with the FC

NSTZuh, 2019-07-17:
I have noticed this as well having just installed my first RadixLi lady

haslinghuis, 2019-10-20:
See https://github.com/betaflight/betaflight/issues/5725
70 degrees warning; 80 degrees critical;
My Radix Li is about 50 degrees (without lipo) and 70 degrees (with lipo) and unify pro race 2 turned on after 10 minutes (just 25mw). Did not fly outside yet, Got no warnings. But with airflow it should be lower. But I am thinking of seperating the vtx from the fc to the topplate.

closed 2022-01-16 by github-actions[bot]