BrainFPV/betaflight #34: Betaflight 4.1

azixaka, 2019-09-29:
Are you planning to release a Betaflight 4.1 based firmware any time soon?

haslinghuis, 2019-10-18:
Looks like schedule is 2 weeks behind original.

azixaka, 2019-10-18:
But it is happening and it's already great! I bought the FC yesterday now waiting for 4.1 as all my quads are on 4.1 and don't even want to try without the RPM filtering. :)

haslinghuis, 2019-10-18:
You can find betaflight 4.1 here: https://github.com/BrainFPV/betaflight/releases. Off course be aware it is not official yet.

pregeth, 2019-10-18:
@azixaka In my opinion, its worth the wait :) Also RC6 is available for testing.

azixaka, 2019-10-18:
I have three quads, three different setups - frames, escs, motors.

I have lots of flights logged with Matek F722 SE, Mini and CL Racing F7 and SP Racing H7 Ex, all on the same track.

It will be so interesting to see how Radix works in the same conditions. Have very high expectations of it as this is the least functional apart from OSD but the most expensive FC on the market. It's also F4 ;)

azixaka, 2019-10-23:
How is it going guys? Where is our 4.1? :)

haslinghuis, 2019-10-23:
Because of special functions (like graphical OSD and Bosch gyro) + having a company (BrainFPV) who wants to have stable tested firmware - we will have to wait a little longer than the original version of Betaflight without those gimmicks. Give it a week or so. But you do not have to wait - it is already here - (see link above) and flash using local file button to your Radix and go?

haslinghuis, 2019-11-07:
@azixaka It is here! You can close this issue now ;-)

azixaka, 2019-11-07:
My Radix is defective, has been RMA'ed just now. I won't get to test it soon what a pity. :)

But yeah, great!

closed 2019-11-07 by azixaka