BrainFPV/betaflight #35: CRSF RSSI not working BF 4.1.0 RC5

crashtest22, 2019-10-04:
LQ/RSSI widget for CRSF seems to not be working as it was in 4.0.x

RADIX full size
TBS Nano 
TBS Evo

Blackdragon4034, 2019-10-05:
It is working for me on 4.1.0 RC5

beeb, 2019-10-05:
I can confirm it's also not working for me. Using the Nano Diversity RX on crossfire 3.0.0

crashtest22, 2019-10-06:
@Blackdragon4034 whats your setup? 
wonder if its different.

SuTeCH007, 2019-10-06:
For me is also working, but I had to do it differently than normal. I was used to set CH8 in BF and on OSD tab set RSSI. But this doesn't work anymore. It will show you the widget, but not the information. I turned off CH8 and RSSI OSD in BF, then with TX and goggles via a menu set Active Element/RSSI (first item in menu) and here you go. Working widget.

beeb, 2019-10-06:
Confirmed, disabling the "AUX" RSSI channel in the Receiver tab of the configurator made the widget work for me too.

SuTeCH007, 2019-10-06:
But the widget is blinking through the flight and also after disarm.

beeb, 2019-10-06:
Yes I also noticed the widget blinking even when LQ is 100% (50Hz) or even 300% (150Hz).

crashtest22, 2019-10-07:
I disabled the rssi in bf receiver tab and can verify it is now working too but flashing all the time

serran0, 2019-10-08:
Confirming same behavior in BF 4.1 RC5, using RadixLi. RSSI showing 0 when AUX channel for RSSI is enabled.

haslinghuis, 2019-10-19:
I never have to enable RSSI on AUX?
Radix Li on BF4.1RC6 with Crossfire Nano (with immortal-T) on v3.21 (X9D+ on OpenTX 2.3.1) it detects battery voltage on main screen and telemetry date. Findings:
1. With older crossfire nano fw 2.95 I got NO DATA. So I updated the crossfire fw on tx [bind] rx [update]. After update the telemetry data came through the telemetry screen.
2. But on this build with old and new crossfire nano fw I have RF signal low/RF signal critical warnings but RQly is 100? Never had this warning. I have 3 quads with crossfire nano.

beeb, 2019-10-21:
@mluessi since you fixed this issue in RC6, do you want to close it?

closed 2022-01-15 by github-actions[bot]