BrainFPV/betaflight #37: RC5 OSD flashing after disarm

beeb, 2019-10-05:
**Describe the bug**
OSD works fine until the quad is disarmed. After disarming, the stats are shown but the OSD disappears and reappears every 1 second. (flashes for a fraction of a second, then is not shown for about 1 second before flashing once more, etc).

**To Reproduce**
Flash RC5 on the Radix, use with Runcam Micro Eagle.

**Expected behavior**
OSD should be constantly shown.

**Flight controller configuration**
```
# 

# diff

# version
# Betaflight / RADIX (RADIX) 4.1.0 Oct  2 2019 / 17:13:08 (4a99bef0e) MSP API: 1.42

# start the command batch
batch start

board_name RADIX

# name: VERT

# resources
resource MOTOR 1 A15
resource MOTOR 4 A02

# feature
feature TELEMETRY

# beeper
beeper -ON_USB

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# serial
serial 0 64 115200 57600 0 115200
serial 2 0 115200 57600 0 115200

# aux
aux 0 0 0 1900 2100 0 0
aux 1 1 1 1300 2100 0 0
aux 2 13 2 1700 2100 0 0
aux 3 35 2 1300 1700 0 0

# master
set dyn_notch_range = LOW
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set dyn_notch_min_hz = 100
set min_check = 1010
set rssi_channel = 8
set dshot_idle_value = 300
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set vbat_min_cell_voltage = 345
set vbat_warning_cell_voltage = 355
set vbat_scale = 118
set ibata_scale = 170
set yaw_motors_reversed = ON
set small_angle = 180
set osd_rssi_alarm = 80
set osd_cap_alarm = 1200
set osd_alt_alarm = 1000
set osd_tim2 = 769
set osd_vbat_pos = 44
set osd_rssi_pos = 14368
set osd_tim_1_pos = 54
set osd_tim_2_pos = 14743
set osd_flymode_pos = 65
set osd_crosshairs_pos = 205
set osd_current_pos = 14720
set osd_mah_drawn_pos = 14688
set osd_craft_name_pos = 362
set osd_warnings_pos = 14697
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_bbox = OFF
set debug_mode = GYRO_SCALED
set gyro_1_sensor_align = CW180FLIP
set gyro_1_align_pitch = 1800
set gyro_1_align_yaw = 1800
set gyro_rpm_notch_harmonics = 1
set brainfpv_osd_font = 2
set brainfpv_altitude_scale = 0
set brainfpv_speed_scale = 0
set brainfpv_map = 0
set brainfpv_show_pilot_logo = 0
set name = VERT

profile 0

rateprofile 0

# rateprofile 0
set roll_rc_rate = 96
set pitch_rc_rate = 96
set yaw_rc_rate = 172
set roll_expo = 9
set pitch_expo = 9
set yaw_expo = 40
set roll_srate = 50
set pitch_srate = 50
set yaw_srate = 49

# end the command batch
batch end

# 
```

**Setup / Versions**
 - Radix full size
 - Crossfire Nano Diversity RX, firmware 3.0.0
 - Runcam Micro Eagle

RoadyFPV, 2019-10-05:
I can confirm this on my Radix. OSD is blinking after disarm.

beeb, 2019-10-05:
Additional information (no idea if it's related). I have a 16GB SD card in there, and no logs appear on there after I fly. Since the BB log number should be displayed in the stats screen I think it's worth mentioning this detail.

crashtest22, 2019-10-07:
I can confirm the same result 
I also had the screen go black for me one time...lasted about 5 seconds....which was about another second or two before I was going to disarm...thank god it came back. JB micro eagle cam. Never had it happen before on older firmware and have at least 100 packs on this build

beeb, 2019-10-07:
> I also had the screen go black for me one time...lasted about 5 seconds....

Was the OSD still showing or completely black? I would not immediately blame the firmware because there are other components that could be responsible (cam if you still had OSD, vTX otherwise).

crashtest22, 2019-10-07:
was full black and for sure understand it may not/probably not be firmware I will keep flying.
if it does it again I will be swapping out the VTX

serran0, 2019-10-08:
Confirming OSD stats flashing/flickering after disarm on BF 4.1 RC5.

KevPV, 2019-10-10:
I can confirm this. Just flashed 4.1 RC5 and the end-stats flash briefly then completely disappear. Pattern repeats every 1-2 seconds. I still have video from the camera, it's just the end stats that disappear. Also running the Runcam Micro Eagle.

beeb, 2019-10-21:
@mluessi since you fixed it in RC6, do you want to close this?

closed 2022-01-15 by github-actions[bot]