BrainFPV/betaflight #38: RC5 SD card blackbox not working

beeb, 2019-10-05:
**Describe the bug**
When inserting a 16GB high quality SD card in the Radix, no log files are created, even though the card is detected and the "FREESPAC.E" file is created on it.

**To Reproduce**
Insert 16GB SD card in the Radix. Check configurator in the Blackbox tab, card is detected and logging is enabled (4kHz). Debug NONE.
Arm quad for a few seconds, disarm.
Notice stats screen displays log number (and increments it each flight).
Look at the SD card contents, there are no files besides the "FREESPAC.E" locked file.

**Expected behavior**
A log file should appear on the SD card for each flight.

**Flight controller configuration**
```
# 

# diff

# version
# Betaflight / RADIX (RADIX) 4.1.0 Oct  2 2019 / 17:13:08 (4a99bef0e) MSP API: 1.42

# start the command batch
batch start

board_name RADIX

# name: VERT

# resources
resource MOTOR 1 A15
resource MOTOR 4 A02

# feature
feature TELEMETRY

# beeper
beeper -ON_USB

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# serial
serial 0 64 115200 57600 0 115200
serial 2 0 115200 57600 0 115200

# aux
aux 0 0 0 1900 2100 0 0
aux 1 1 1 1300 2100 0 0
aux 2 13 2 1700 2100 0 0
aux 3 35 2 1300 1700 0 0

# master
set dyn_notch_range = LOW
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set dyn_notch_min_hz = 100
set min_check = 1010
set rssi_channel = 8
set dshot_idle_value = 300
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set vbat_min_cell_voltage = 345
set vbat_warning_cell_voltage = 355
set vbat_scale = 118
set ibata_scale = 170
set yaw_motors_reversed = ON
set small_angle = 180
set osd_rssi_alarm = 80
set osd_cap_alarm = 1200
set osd_alt_alarm = 1000
set osd_tim2 = 769
set osd_vbat_pos = 44
set osd_rssi_pos = 14368
set osd_tim_1_pos = 54
set osd_tim_2_pos = 14743
set osd_flymode_pos = 65
set osd_crosshairs_pos = 205
set osd_current_pos = 14720
set osd_mah_drawn_pos = 14688
set osd_craft_name_pos = 362
set osd_warnings_pos = 14697
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_bbox = OFF
set debug_mode = GYRO_SCALED
set gyro_1_sensor_align = CW180FLIP
set gyro_1_align_pitch = 1800
set gyro_1_align_yaw = 1800
set gyro_rpm_notch_harmonics = 1
set brainfpv_osd_font = 2
set brainfpv_altitude_scale = 0
set brainfpv_speed_scale = 0
set brainfpv_map = 0
set brainfpv_show_pilot_logo = 0
set name = VERT

profile 0

rateprofile 0

# rateprofile 0
set roll_rc_rate = 96
set pitch_rc_rate = 96
set yaw_rc_rate = 172
set roll_expo = 9
set pitch_expo = 9
set yaw_expo = 40
set roll_srate = 50
set pitch_srate = 50
set yaw_srate = 49

# end the command batch
batch end

# 
```

**Setup / Versions**
 - Radix on BF 4.1.0 RC5
 - SanDisk Ultra SD Card 16GB

azixaka, 2019-10-28:
Is it fixed for you in RC6?

beeb, 2019-12-26:
This problem is not fixed in 4.1.1. I just tried with a reputable SanDisk Ultra 16Go card and don't get a log on the card, although I can totally read and write to it even though USB Mass Storage on the FC.

closed 2022-01-15 by github-actions[bot]