BrainFPV/betaflight #40: broken build

bubi-007, 2019-10-19:
the current build is broken.

%% (speed optimised) ./src/main/drivers/accgyro/accgyro_spi_bmi160.c
make[2]: *** Keine Regel vorhanden, um das Ziel „src/link/stm32_flash_f446_chibios.ld“,
  benötigt von „obj/betaflight_v4.1.0-RC6-BRAIN-1_BRAINRE1.elf“, zu erstellen.  Schluss.
make[2]: *** Es wird auf noch nicht beendete Prozesse gewartet....
%% (size optimised) ./src/main/drivers/barometer/barometer_bmp280.c
make[2]: Verzeichnis „/home/rob/brainfpv“ wird verlassen
make[1]: *** [Makefile:561: binary] Fehler 2
make[1]: Verzeichnis „/home/rob/brainfpv“ wird verlassen
make: *** [Makefile:493: BRAINRE1] Fehler 2

justinflipflops, 2021-05-13:
This should be closed the most recent version builds just fine.

robert-b, 2021-05-14:
wow - just in time ...

justinflipflops, 2021-05-14:
> wow - just in time ...

I didn't notice the date. No need to be a prick though, Kudos!

robert-b, 2021-05-14:
for me it works as well.
but it can be closed as you where mentioning it.

closed 2022-01-15 by github-actions[bot]