BrainFPV/betaflight #41: RSSI not working in OSD Brain Radix 4.1 RC 6

Mrd1114, 2019-10-21:
**Describe the bug**
The RSSI/LQ display in the OSD is not working.  It flashes intermittently.  It was working before the flash, and I've confirmed the RSSI is on the proper AUX channel.  The crossfire receiver is configured properly, handling RSSI on the proper channel.

**To Reproduce**
Flash betaflight firmware 4.1 release candidate 6 onto a BrainFPV Radix, and check OSD display after configuration.  I've flashed it to two separate boards in two separate frames.  Both are displaying this issue.

**Expected behavior**
Expected RSSI to update and display actual values instead of a zeros and -20 dB

**Setup / Versions**
 - Flight controller BRAINFPV Radix 30 x 30 board
 - Other components TBS crossfire nano, TBS Unify Pro HV

SuTeCH007, 2019-10-21:
Turn off your AUX in BF, and turn on your widget and RSSI via OSD.

Mrd1114, 2019-10-21:
I will try that tonight, thank you.

Mrd1114, 2019-10-22:
Yeah, that worked perfectly for one build, and not for the other.  The problem build is, of course, also using a Radix board, the only difference between them is one is using a Crossfire Nano receiver, and the other (the one that the problem persists on) is using the TBS micro receiver V2.  I've tried re-flashing with release candidate 6, verified that I'm getting RSSI on the telemetry (using a FrSky Taranis with Open TX 2.2).  Could there be a firmware issue on the receiver?

SuTeCH007, 2019-10-22:
> Yeah, that worked perfectly for one build, and not for the other. The problem build is, of course, also using a Radix board, the only difference between them is one is using a Crossfire Nano [...]

Do you have the last firmware in your CRSF?

Mrd1114, 2019-10-22:
I updated the TBS Crossfire firmware on the TX and the Micro receiver.  The RSSI still isn't working, and the widget is just blinking.  Really strange.  It worked perfectly for one rig but not this one.  I've tried re-flashing Betaflight 4.1 RC 6 again, same result.

haslinghuis, 2019-10-28:
I confirm RSSI Value (LQ 300) is displayed in OSD (Radix Li with BF4.1 RC6) - I had to update Crossfire TX firmware and rebind (+ update) my receivers.

Mrd1114, 2019-12-03:
Update:

Yeah, this was all caused by being on TBS Firmware 2.42. I just found out that TBS switched from TBS Agent to TBS Agent X. I was still using the former, and it was telling me my firmware was up to date, when its all the way up to 3.43. By updating, it completely fixed the issue. I really wish the TBS Agent would have notified me somehow that it was obsolete and to switch to TBS Agent X. Would have saved me a lot of heartache. I'm closing the issue.

closed 2019-12-03 by Mrd1114